mck1117/proteus #57: VR polarity minor mess

rusefillc, 2020-07-23:
[image]

mck1117, 2020-07-28:
While this wasn't intentional, with the polarity flipped, the behavior actually matches that of microRusefi.  The TLE8888 is a non-inverting setup, while the MAX992x is inverting, so flipping the pins makes them both behave the same way.

See both datasheets (TLE on top, MAX on bottom):
[image]

Besides, it's easy to switch the polarity in software.

So while technically wrong, I'm going to leave it wrong.

mck1117, 2020-07-28:
Comment added on the schematic explaining the logic above, with a link to this issue.

closed 2020-07-28 by mck1117

rusefillc, 2020-07-28:
Maybe a comment on the schematics confirming that this is now a feature?