mck1117/proteus #58: Proteus ETB1 MR capacitor damage

ErichPopp, 2020-07-25:
With no grounds connected to the black 23pin connector and while powered on, the black 35 pin connector was disconnected and reconnected. When it was connected again the capacitor failed and the trace between the MR pin and capacitor was vaporized.

ErichPopp, 2020-07-25:
[image]
[image]

mck1117, 2020-07-25:
Does the cap measure as a short circuit?

ErichPopp, 2020-07-25:
I ended up not getting back out there yesterday. It’s my birthday weekend
so I may not be out there again until Monday. I’ll let you know as soon as
I can.

mck1117, 2020-07-25:
No problem!

rusefillc, 2020-07-26:
Happy Birthday @ErichPopp!

mck1117, 2020-08-07:
@ErichPopp have a chance to test that cap?

mck1117, 2020-11-03:
I seem to remember the cap ended being a short circuit, probably due to a crack in it from the proximity to the connector.  In case it was a crack due to overvoltage (inductive spike), an electrolytic was added to the `12v_MR` net in 0.4 as extra insurance, in addition to the ceramic near each ETB driver.

closed 2020-11-03 by mck1117