mck1117/proteus #67: Use Passat B6 ECU case and connector

rusefillc, 2020-09-07:
06 2006 Volkswagen Passat 2.0L ECM ECU Engine Computer Module | 3C0 907 115 G case is available from China brand new. also used from junk yards.

crimped pigtails are available from junk yard.

would it be the simplest way to give amazing Proteus an amazing case?

rusefillc, 2020-09-07:
[image]

https://www.ebay.com/itm/154-pin-ECU-Connector-Bosch-Type-for-Patch-Harness-Audi-Volkswagen/333606651255

rusefillc, 2020-09-07:
[image]

mck1117, 2020-09-12:
154 pins is a lot - maybe some other features could get added, like integrated dumb coil drivers (igbt)

rusefillc, 2020-09-12:
is now a probability of Proteus moving in this direction?

I can make the PCB outline and card board test it in a few weeks unless we already have the outline?

rusefillc, 2020-09-26:
https://github.com/rusefi/rusefi_documentation/blob/master/oem_docs/cases/BS616%20New%20Harvard%20Cover.dwg could be the case

rusefillc, 2020-10-05:
[
[image]

[image]

rusefillc, 2020-11-22:
board outline https://github.com/mck1117/proteus/pull/85

mck1117, 2021-07-18:
https://github.com/mck1117/atlas

closed 2021-07-18 by mck1117