mck1117/proteus #93: VNLD5160 no avilable at all

NoTwixD, 2021-02-22:
45 weeks on mouser 
[image]
[image]

NoTwixD, 2021-02-22:
VN7 series are only single channel in so-8 Package

NoTwixD, 2021-02-22:
VNLD5090-E more expensive higher Drain current up to 13 A (yehh i know its overkill) lower On resistance 160>80mOhm its mean less heat ;)

NoTwixD, 2021-02-22:
and one older VNS3NV04DP-E not plug and play

NoTwixD, 2021-02-22:
same story with BTS4175SGA

NoTwixD, 2021-02-22:
I belive the tls115 i last one realy missing part of proteus at the moment

mck1117, 2021-03-28:
Yes, the 5090 is interchangeable with the 5160, and while it does dissipate less heat, the current limit is also higher (which you may or may not want).

BTS4175 is interchangeable with the BSP762 and ISP762.

mck1117, 2021-03-28:
The world is having a silicon shortage right now, so this isn't really a Proteus issue.  It's just going to be a mess for a while.

closed 2021-03-28 by mck1117