mck1117/proteus #104: Schematics note which diodes are usually populated which ones are not

rusefillc, 2021-10-02:
Schematics shows

[image]

looking it seem like proteus has 4 diodes. In reality two diodes on top side are always populated while two diodes on the back are pretty much always NOT populated

let's a have a note of that on the schematics?

rusefillc, 2022-04-02:
also text all over text is an eye sore

mck1117, 2022-04-03:
[image]

What if we just fit all 4 instead :)

rusefillc, 2022-04-03:
off topic: 

I have sad news for you RE

[image]

closed 2022-07-07 by mck1117