mck1117/proteus #105: 0.4 unit not powering up from +12v IGN

rusefillc, 2021-10-31:
I believe I have a damaged Proteus 0.4 board

I apply +12v on IGN pin 18, I have GND and I do not get +5

@qwerty-off suggested that I check voltage on the huge diode (?)

I have +12 on one side of huge D but not on the other

[image]

do we suspect damaged huge D? how would that happen?

Qwerty-OFF, 2021-10-31:
My mistake. we should check this diode
[image]

rusefillc, 2021-11-01:
both 12v points measure +12 - both left side of diode from top pic and right side of capacitor

right side of inductor is a huge LOL: applying "voltage" mode DMM there TURNS ON STM32. and measures +5v

rusefillc, 2021-11-01:
Adding solder on both sides of inductor has resolved this issue

this 0.4 is from Jan 2021 batch when JLC did not populate inductors

closed 2021-11-01 by rusefillc