mck1117/proteus #107: chip shortage: LINEAR VOLTAGE REGULATOR

rusefillc, 2021-11-16:
TLS115D0EJXUMA1  only available for huge MOQ, do we have alternatives?

Qwerty-OFF, 2021-11-22:
TLS115B0EJXUMA1
Similar, but without PG output
[image]

rusefillc, 2021-11-22:
thank you @Qwerty-OFF I've placed the order

[image]

rusefillc, 2021-11-24:
[image]

rusefillc, 2021-12-01:
hellen has a somewhat similar drama with https://www.mouser.com/ProductDetail/Texas-Instruments/LMR14020SSQDDAQ1?qs=sGAEpiMZZMsyYdr3R27aV1%2FsE%2F4eX1bpBt4%2FHWGByuc%3D

are those compatible or not?

mck1117, 2021-12-01:
unfortunately not really related - TLS115 is for the sensor supplies, and Proteus ALSO has an LMR14020 for the main supply.

rusefillc, 2021-12-03:
Mouser has shipped
I've also ordered 386 of TLS115D0EJXUMA1

rusefillc, 2021-12-03:
See also https://github.com/andreika-git/hellen-one/issues/119

rusefillc, 2022-04-02:
I have a pile of both TLS115  and LMR14020

closed 2022-04-02 by rusefillc