mck1117/proteus #126: proteus HW0.7v dont have suppressor diode on ETB Power input

WOOL929, 2024-02-08:
On the Proteus v0.7 board, the tle9201 power input does not have a suppressor Diode.

[image]

It seems that if the top pair of arms fails, then the reverse current from the throttle return could become a problem for the nodes on the 12v_MR bus?!

[image]

The datasheet has it.
[image]

Maybe it's not even a problem? I'm trying myself as a bug reporter

rusefillc, 2024-02-08:
@WOOL929 did you get some proteus damage which got you to look into this or is this just theoretical?

WOOL929, 2024-02-08:
> @WOOL929 did you get some proteus damage which got you to look into this or is this just theoretical?

not in my case.  just a theory.  in my case the culprit is c1203 and out- on the ground.  I have already fixed this.

rusefillc, 2024-02-09:
https://github.com/mck1117/proteus/issues/127